bblacey/MySensorIRBlaster #9: 5VDC vs. 3.3VDC for IR LED circuit

bblacey, 2015-07-07:
The v0.5 version of the IR LED circuit was 5V requiring 35 ohm 0.35watt current limiting resistors and was laid out using 0603 resistor footprints.  During part selection, I found that it was nearly impossible to find a resistor that could meet the power requirements so for the dev board, I reverted to driving the IR LEDs with 3.3v allowing the use of a 15 ohm 0.18 watt resistor in the 0603 package. 

The open issue is which voltage should be used for the final board.  If 5VDC, then at a minimum, we need to change the resistor footprint to 0805.

tbowmo, 2015-07-08:
If you could put in an extra led,  so you have 8 in total,  then put  2 in series with a resistor,  multiply with 4. Will also lower the current through the FET.

bblacey, 2015-07-08:
Good idea Thomas.  What if I used 3-pair plus one single?  I would still have 4x100mA current draw through the FET and I think I would only need to use slightly different current limiting resistors for the 1 vs. pairs to do it.  In fact, the IR LED FV is 1.4-1.8 at 100mA so I would only need to offset from 3.3 down to 2.8-3.6 so maybe I wouldn't even need a current limiting resistor for the pairs. Do you see any significant issues with that approach?  The IR design uses an odd number of LEDs so that one of them is completely forward-facing at a 0-degree angle for maximum range.

Another advantage for using 3.3VDC for the IR LED circuit is that if there are two LDOs, one won't need to dissipate a larger voltage differential between the power supply and the 3.3VDC LDO and users would be able to use a more common 5VDC power supply.  Taking it a step further would be to consolidate to a single LDO (the AMS1117 can push 1A) with additional capacitance to compensate and push all components to the top-side while still keeping the board under 5x5cm...  Good idea/bad idea?

tbowmo, 2015-07-08:
Hmm of course you could just have 7 leds, with a larger resistor for the
lonely led.

Could probably also go with a single ldo. But definitely need  good
decoupling so the 38khz carrier frequency doesn't mess up the nrf24 module,
or other things on board. I think you should make some prototype to
evaluate the led circuitry. And check if resistors can be left out.

Good idea Thomas. What if I used 3-pair plus one single? I would still have
4x100mA current draw through the FET and I think I would only need to use
slightly different current limiting resistors for the 1 vs. pairs to do it.
In fact, the IR LED FV is 1.4-1.8 at 100mA so I would only need to offset
from 3.3 down to 2.8-3.6 so maybe I wouldn't even need a current limiting
resistor for the pairs. Do you see any significant issues with that
approach? The IR design uses an odd number of LEDs so that one of them is
completely forward-facing at a 0-degree angle for maximum range.

Another advantage for using 3.3VDC for the IR LED circuit is that if there
are two LDOs, one won't need to dissipate a larger voltage differential
between the power supply and the 3.3VDC LDO and users would be able to use
a more common 5VDC power supply. Taking it a step further would be to
consolidate to a single LDO (the AMS1117 can push 1A) with additional
capacitance to compensate and push all components to the top-side while
still keeping the board under 5x5cm... Good idea/bad idea?

bblacey, 2015-07-09:
Interesting test of the IR LEDs.
The datasheet states typical VF is 1.2 for IF 20mA.
And 1.4 for 100mA.
I have one connected to a bench supply.  

* At 1.0 VDC, it is drawing 26mA
* 1.1 = 50mA
* 1.2 = 82mA
* 1.3 = 155mA
* 1.4 = whopping 260mA!

Far cry from the data sheet. So pairing is a good idea unless we use a heavy duty FET 😉I should try a few to see the variance between individual LEDs.
And my bench supply isn't the most accurate.  I should actually use a couple of meters to measure voltage and current separately from the bench supply.