bblacey/MySensorIRBlaster #11: Evaluate using the SAM D2x series MCU in lieu of the 328p & companion 512Kb flash for MySensors Core

bblacey, 2015-07-09:
The ATMEL D2x series MCU seems like a good candidate for the MySensors MCU core because it is equipped with up to 256KB of on-board flash for program memory and 32KB RAM for data - the increased flash might be sufficient for OTA allowing us to reduce the total part count.  The SAM D2x series uses a 32-bit ARM Cortex M0 CPU that can be clocked to 48MHz.

* D20G18A-AU is a good Sensor/Actuator candidate and can be purchased for as low as $2.44/ea - http://www.newark.com/atmel/atsamd20g18a-au/microcontroller-mcu-32-bit-cortex/dp/19X7616
* Comes in a TQFP-48 package that is physically the same dimensions as the 328p but packs 16 more pins in by adding 4 pins/side at a 0.05mm pitch (versus 328p 0.08mm pitch)
* Operating voltage 1.62V - 3.63V
* Power consumption: 8uA - 70uA/MHz (idle and standby sleep modes, sleepWalk peripherals)
* 256KB flash, 32KB RAM
* 38 GPIO pins
* 6 Timer Counters that can be used for PWM
* Two-pin SWD programming, test and debug :+1: 
* Brown out detection and Watchdog

The Arduino Zero uses the D21E that adds USB.

[image]

@tbowmo and @fallberg, it would be great if you guys would weigh in on this...  Seems like an interesting option but what do you guys think?

tbowmo, 2015-07-09:
Hmm sounds interesting. But is all software directly compatible with it? (of course with appropriate  compiler) 

Next question. What programmer is needed to do initial programming? Can it be reprogrammed via USB? Thinking about end users here, do they need to buy new hardware to use it with their own sketch? 
Can we do Ota updates easily?

tbowmo, 2015-07-09:
Been looking at the schematic of arduino zero, it's a two processor design, implementing an atmega32u4 as a programming device for the SAM. an jtagice3 could also be used to programming the SAM.

At a first glance, this means that the end user is not able to program this device easily with arduino, unless we also implement the atmega32u4 on board to handle things. Or the end user needs to purchase a new programming device.

I haven't looked deeply into the datasheets yet, so this is just from an initial browsing on documents, both on arduino zero, and on atmel SAM D20 feature leaflets. So I might be wrong in my assumptions

bblacey, 2015-07-09:
Peter Easton, the author of ControLeo pointed me to the [Neutrino - an Arduino Zero compatible kickstarter project](https://www.kickstarter.com/projects/rabidprototypes/neutrino-the-tiny-32-bit-arduino-zero-compatible/description) (He is planning to use the Atmel SAM for his next project).  The schematic is below and seems pretty straight-forward.

[Schematic page 1](http://rabidprototypes.com/Neutrino/Neutrino-150-Schematic-1.png)
[Schematic page 2](http://rabidprototypes.com/Neutrino/Neutrino-150-Schematic-2.png)

Also, if we went this way, we would probably want to go with the D21G18A-AU for maximum Arduino IDE and library compatibility.  Mouser has D21G18A-AU for $4.56 in quantities of 10.

I used the Arduino Board Manager to add support for the SAMD Arduino and then looked at the boards.txt section - looks pretty straight-forward:

```
arduino_zero_edbg.name=Arduino Zero (Programming Port)
arduino_zero_edbg.vid.0=0x03eb
arduino_zero_edbg.pid.0=0x2157
arduino_zero_edbg.upload.tool=openocd
arduino_zero_edbg.upload.protocol=sam-ba
arduino_zero_edbg.upload.maximum_size=262144
arduino_zero_edbg.upload.use_1200bps_touch=false
arduino_zero_edbg.upload.wait_for_upload_port=false
arduino_zero_edbg.upload.native_usb=false
arduino_zero_edbg.build.mcu=cortex-m0plus
arduino_zero_edbg.build.f_cpu=48000000L
arduino_zero_edbg.build.usb_product="Arduino Zero"
arduino_zero_edbg.build.usb_manufacturer="Arduino LLC"
arduino_zero_edbg.build.board=SAMD_ZERO
arduino_zero_edbg.build.core=arduino
arduino_zero_edbg.build.extra_flags=-D__SAMD21G18A__ {build.usb_flags}
arduino_zero_edbg.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld
arduino_zero_edbg.build.openocdscript=openocd_scripts/arduino_zero.cfg
arduino_zero_edbg.build.variant=arduino_zero
arduino_zero_edbg.build.variant_system_lib=
arduino_zero_edbg.build.vid=0x2341
arduino_zero_edbg.build.pid=0x804d
arduino_zero_edbg.bootloader.tool=openocd
arduino_zero_edbg.bootloader.file=zero/samd21_sam_ba.bin
```

Should use the same programmers, etc. but need to verify.  Intriguing to say the least...  I would like to hear from @henrikekblad if 64K words (32-bit word size) is sufficient for a sketch and an OTA update.  The current 328p design has 32K words (8-bit word size) on board plus the 512 Kb (64K words) off-board  so how much do we need for OTA updates and will it work?  Sketches would need to stay below 32K words I suspect.

tbowmo, 2015-07-09:
One thought, is the libraries that we need ported? Can we use the arduino DUE (It's using an cortex m3) as a mysensors gateway?

I think we need to have a POC running with the processor, a radio module (ideally both nrf24 and rfm69) and check that we can communicate with the sensor network successfully.. Also check that we can use w5100 / esp8266 for wired / wireless "internet" access

One part I really like about it, is that it's running on 3v3.. So I can skip all level converters, and run everything on a single vcc rail in the GW.

bblacey, 2015-07-09:
I have ordered two [Neutrinos](http://rabidprototypes.com/product/neutrino/) for $20/each and also saw that you can buy an a [SAM D21 Xplained Pro Evaluation Kit](http://www.atmel.com/tools/ATSAMD21-XPRO.aspx) for $42.65 from ATMEL - of course, there is always the [Arduino Zero](http://store-usa.arduino.cc/products/abx00003) for $50 but it probably contains more "parts" than what we want to use as a reference.  Both boards come with 256KB flash so they each would be good platforms for experimentation.

Emmanul123456, 2017-01-06:
@bblacey 
hello friend ,I am a beginner in programming, and I am using SparkFun SAMD21 Dev Breakout, I am enthusiastic since you mention uploading sketch through OTA, I would like to know if you accomplished your task, I have a school project and I would like to implement it through GPRS, you could mention me as I can get to this, thank you.

bblacey, 2017-01-07:
@Emmanul123456, the new MySensors gateway uses the SAMD21 processor but does not require OTA.  @tbowmo designed and implemented the MySensors gateway and is also working on a [major revision to his original sensebender micro sensor](https://forum.mysensors.org/topic/5329/sensebender-micro-mk2/12) that will also use a SAMD processor.  OTA is performed by the bootloader and currently the [MYSBootloader](https://forum.mysensors.org/topic/5665/mysbootloader-1-3-0-beta-3/4) that currently runs on the ATMEL, ATMEGA AVR MCUs. @tekka, is the core member for the MYSBOOTLOADER and has  not yet had time to port develop OTA support for the SAMD line of MCUs.  I suggest that you post these questions to the [MySensors forums](https://forum.mysensors.org/category/13/development) in order to get better support because all the MySensors core members and the vibrant community members are active on the forums and can advise and help you with your project.  Good luck with your project and welcome to MySensors!