bblacey/MySensorIRBlaster #15: Constant-Current IR LED circuit for maximum IR luminance?

bblacey, 2015-07-13:
Compare with v0.5 parallel IR LEDs driven by a high-current FET.  At 3.3VDC, will need to drive 3 pair with 1 single.  5 VDC?

[Sharp IR LED Circuits](http://www.analysir.com/blog/wp-content/uploads/2013/11/Sharp-IR-LED-Circuits.pdf?5b4871_
[Constant current infrared LED emitter circuit](http://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

/cc @fallberg

bblacey, 2015-07-13:
[FET/NPN constant current](http://www.instructables.com/id/Power-LED-s---simplest-light-with-constant-current/)
[TI constant current LED driver](http://www.ti.com/product/tlc59108/description)
[ST constant current LED driver](http://www.st.com/web/en/resource/technical/document/datasheet/CD00156241.pdf)

bblacey, 2015-07-13:
@fallberg, the more I read about this, the more I am convinced that putting the IR LEDs on a constant current circuit is the way to go.  I haven't done all the math but we can completely separate the MCU IC core on 3.3VDC from the IR LEDs and use the sense resister to compensate for varying input voltages.  So, we can drive the board with up to 12V (AMS1117 max VDC), and use an non-voltage regulated constant-current circuit to drive the LEDs.  I have included some short reads on the subject - including cheap DIY circuits that use a FET for switching and a sense resistor and NPN transistor for the current regulation...  Check them out and let me know your thoughts.

tbowmo, 2015-07-13:
Hmm have actually been thinking about the constant current, but am a bit uncertain about the switching circuit. 

You can probably use an 1117 as CC source,  (you can with an lm317)

bblacey, 2015-07-13:
@tbowmo thanks for the insight on the lm317 as a CC source - we should be able to switch with a FET after the CC regulator, right?  Is your uncertainty about circuit integrity?  

That said, it is clear that we have a number of options (VR as CC, NPN pair w/resistor, FET/NPN pair w/resistor, or some CC LED driver).  I guess in the end we will need to consider part count and cost - the odd number of LEDs may sway me towards a driver such as the [CAT 4101 LED Driver with PWM control](http://www.onsemi.com/pub_link/Collateral/CAT4101-D.PDF) that can take 3-6VDC and drive in-series LEDs up to 1A (we need 700mA)...  This particular part is a bit bigger and more expensive than what we may want but a single package for an array of in-series LEDs with constant current is and built-in switching is very attractive...  With a bit of searching, we might be able to find a similar part that is smaller and less expensive.

bblacey, 2015-07-13:
Another TI argument against parallel LEDs...

>Pitfalls of Parallel LED Arrays
Whenever LEDs are placed in parallel, the potential exists for a mismatch in the current that flows through the different branches. The forward voltage, VF, of each LED varies with process, so unless each LED is binned or selected to match VF, the LED or LED string with the lowest total forward voltage will draw the most current (Figure 4). This problem is compounded by the negative temperature coefficient of LEDs (and all PN junction diodes). The LEDs that draw the most current suffer the greatest increase in die temperature. As their die temperature increases, their VF decreases, creating a positive feedback loop. Elevated die temperature both reduces the light output and decreases the lifetime of the LEDs.

Reference: http://www.ti.com/lit/an/snva253a/snva253a.pdf

fallberg, 2015-07-13:
A cheap LED driver circuit would probably be the best option since it reduces both the number of necessary components, provides a tested environment (within specifications) and (I suppose) cost less PCB area to implement.